Chris--A/Keypad #4: Library conflict-GSM-Keypad, still needs to be fixed! #4858

DouglasBair, 2016-04-24:
Take a blank Arduino sketch, include and . try to compile. I receive this error:
"C:\Program Files (x86)\Arduino\libraries\GSM\src/GSM3MobileAccessProvider.h:37:36: note: previous declaration 'GSM3_NetworkStatus_t IDLE'
enum GSM3_NetworkStatus_t { ERROR, IDLE, CONNECTING, GSM_READY, GPRS_READY, TRANSPARENT_CONNECTED, OFF};"

Researching this, I see many years of users asking that this be fixed, and not expecting to be told 
fix it yourself by editing a library. What then for the next release, fix it again? 
I've already check the forum and the issues. Many 'smart alecs' say its fixed, where?
This happens in v1.6.5, 1.6.6, and in (.org's) 1.7.9.

When will Arduino.cc fix this?

And, yes, I'm a newbie. So if I'm expecting too much or have entered the lib names incorrectly,
let me know.
Doug

Chris--A, 2016-04-24:
Hi, this error is new to me!  (this isn't my lib, I just created a repo so it can be used through the Arduino library manager). However I have taken control of its updates and now take responsibility, so I thank you for raising the issue.

This looks like a simple error involving multiple names.
Its not specific to this lib or the GSM lib.

The usual way for this to be resolved in C++ is to use namespaces by wrapping the include of one of the libs in a namespace giving it an extra level of indirection. Or as others may have mentioned, pick one of the libraries and change the name so there is no conflict (not what you want).

> Researching this, I see many years of users asking that this be fixed, and not expecting to be told 
fix it yourself by editing a library. What then for the next release, fix it again? 

Here is how you can implement it so you will not have to change any library code with future updates. And yes you can be the first to have a working solution without touching the library!

First wrap the include in a namespace in your sketch file:

```C++
namespace KP{   //You can use any name you like (as long as it is unique)
  #include <Keypad.h>
};
```

Next step, create a cpp file in your sketch directory (from the IDE use the menu `Sketch->Show Sketch folder`).

Inside the new cpp file add this code:

```C++
namespace KP{  //Same namespace name you used in your sketch
  #include <Key.cpp>
  #include <Keypad.cpp>
};
```

Now in your sketch, you need to access the library components with the fully qualified name. Here is a sample:

```C++

namespace KP{
  #include <Keypad.h>
};

const byte ROWS = 4; //four rows
const byte COLS = 3; //three columns
char keys[ROWS][COLS] = {
{'1','2','3'},
{'4','5','6'},
{'7','8','9'},
{'*','0','#'}
};
byte rowPins[ROWS] = {5, 4, 3, 2}; //connect to the row pinouts of the kpd
byte colPins[COLS] = {8, 7, 6}; //connect to the column pinouts of the kpd

KP::Keypad kpd = KP::Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );

void setup() {}

void loop() {
  switch (kpd.key[0].kstate) {  // Report active key state : IDLE, PRESSED, HOLD, or RELEASED
      case KP::PRESSED:

        break;
      case KP::HOLD:

        break;
      case KP::RELEASED:

        break;
      case KP::IDLE:
        
        break;
  }
}
```

Notice the `KP::` in front of the Keypad library components. Also `#define`s aren't specific to any namespace so `makeKeymap` does not need the extra indirection.

Resolving these conflicts is not an easy task, even for people familiar with C++, which probably is why there is no real solution to the problem (until now!).

Let me know how you go, and I'll help you through it if you get stuck.

DouglasBair, 2016-04-24:
Chris--A,
I thank you for your effort.  This may fix my problem but it reveals the fact that who ever is running
the show (Arduino.cc) has dropped the ball.  Both libs are included in each release of the IDE.  It seems that no one is testing compatablity of the libs.  Even if I make the change that you offer, the concept/idea of having a standard/common library is defeated.
Please pass this on to those in charge or tell me where to send it.

DouglasBair, 2016-04-24:
Hi Chris,
Thanks for such a quick response.  I did a reply on Getup, not sure if I did
it correctly,
so here is a copy of it;
Chris--A,
I thank you for your effort. This may fix my problem but it reveals the fact
that who ever is running
the show (Adriano.cc) has dropped the ball. Both labs are included in each
release of the IDE. It seems that no one is testing compatibility of the
labs. Even if I make the change that you offer, the concept/idea of having a
standard/common library is defeated.
Please pass this on to those in charge or tell me where to send it.
 
I thought I'd mention that Ad fruit has a GEM shield and a keypad project
the call a phone.
They call their GSM library 'FONA'.  It doesn't have any conflicts.  I
wonder if it is of any use to
be used instead of the current one.  I know it uses a different GSM module.
Thanks again,
Doug

  _____  

be fixed! #4858 (#4)

<https://github.com/Chris--A/Keypad/issues/4#issuecomment-213909336> it on
GitHub
<https://github.com/notifications/beacon/ARmIyD2JeWvQ-2ll5B3VkTFK0qnrTx-6ks5
p6yGogaJpZM4IOUJj.gif> 

  _____  

No virus found in this message.
Checked by AVG - www.avg.com
Version: 2016.0.7539 / Virus Database: 4556/12091 - Release Date: 04/24/16

Chris--A, 2016-04-24:
I can pass it on, but it probably wont do much. I didn't know it was the Arduino GSM library you were using, when searching the code you posted it appeared in many other repos.

> Even if I make the change that you offer, the concept/idea of having a
standard/common library is defeated.

There is no fault from the Arduino team here. The keypad library is one of thousands available through the library manager and the random internet. You can't expect them to try every single possibility for a conflict.

Which is why you can do as I posted, which is a standard method of resolving these things.

Changing this library now will break a lot of working code found in peoples projects and tutorials for others. And is probably the same for the GSM library.

Unfortunately you will either have to do the fix that I posted, or edit one of the libraries. Its probably easier to just edit this Keypad library, it is not going to change much if at all in the near future.

DouglasBair, 2016-04-25:
Hi Chris,
 
I maintain that it is the responsability of the person or team that bundles
the code and libraries 
that make the IDE installation package.
 
If I sent you an electronics kit that ran on 3.3VDC and included a 12VAC
supply, whose fault would
it be if you put it together and it caught on fire?
 
Someone gathered the components together and created the EXE and ZIP files
for mass distribution world wide.
 
Remember, out of 100 users (of anything) only about 20 will go to the effort
to mention a problem.
 
I wonder how many other conflicts there are that we don't know about.  Also,
I had to hunt for where
to post a problem and then post it.  I waited 9 days before investigating
how to make my post get a response.
It didn't get any response, I accidentaly found your name associated with
some other problem/post and decided
to try sending it to you.   How many others have said 'screw it!' and gave
up?
 
Tell me where to send my concerns (something else a professional company
would show).
Microsoft, Intel, Atmel, and thousands of others have their support address
located on their site and it isn't at some other site.
Also, Github doesn't show anything saying send, post, email here for Arduino
support.
 
You comment "..... it is not going to change much if at all in the near
future." is the .cc group doomed by the .org?
 
Doug
 
  _____  

be fixed! #4858 (#4)

<https://github.com/notifications/beacon/ARmIyKPOV2yy2P7TCJQlNgrOAnVINc_pks5
p6_j_gaJpZM4IOUJj.gif> 

  _____  

Version: 2016.0.7539 / Virus Database: 4563/12099 - Release Date: 04/25/16

Chris--A, 2016-04-25:
> You comment "..... it is not going to change much if at all in the near
future." is the .cc group doomed by the .org?

Neither group has control over this repository.

DouglasBair, 2016-07-15:
The startup banner on the Arduino IDE program says "Created and supported by Arduino.cc".
Both libraries (GSM.h and Keypad.h) are provided within the IDE load.   I can't think of anyone else 
to assign responsibility towards.
I have built several programs with IDE version 1.6.6. When I upgrade to newer versions I find things that won't work anymore.  Who do I announce this to?  I am having to test each IDE to see what is broken 
and which programs are affected.  Should I start another issue?  I would believe that the owners would
want users (people buying boards) to have faith that IDEs are being regression (backward compatible)
tested.  I am not bitching on you, you have very helpful.

DouglasBair, 2016-07-15:
I accidentally closed issue.  I don't see this fixed by any new library.  Please post where it is fixed.
To ask me to modify code does not fix this, it just lets someone else encounter the same problem.
Then, they spend many frustrating hours wondering WTF?

Chris--A, 2016-07-16:
> The startup banner on the Arduino IDE program says "Created and supported by Arduino.cc".
Both libraries (GSM.h and Keypad.h) are provided within the IDE load

No, this library is provided through the library manager, which are libraries not provided by the IDE. Anyone can write a library and add it to the library manager.

If you find bugs or improvements for the IDE, then the Arduino folk are the ones you need to talk to. That banner is for the IDE software, not the thousands of libraries that hundreds of users have added to the library manager.

I have posted a fix for this, where you do not have to modify the library here: https://github.com/Chris--A/Keypad/issues/4#issuecomment-213909336

The fact is this library is not going to change. The original Keypad library is a part of the Wiring library, and is far older than Arduino, and its GSM lib. There are far too many tutorials and references which will be made incorrect with the change you want. And I do not have access to most of them, let alone have the time to update the entire internet.

There are far more uses of the Keypad library than there are using the GSM lib. And for the odd few who are, they can either modify the library, or use the perfectly acceptable fix I have proposed above.

If you want to remove the duplication of the `IDLE` constant, your only port of call is the GSM library. And changing the code there will affect far less people.

DouglasBair, 2016-07-16:
I believe your idea of altering the GSM library is better than the wrapper method for me.
I wondeer if there is a keypad library out there somewhere that is compatible.
I know that Adafruit shows using both the keypad and their GSM board together to make a phone.
I guess I'll buy their GSM board to make my alarm system.
I'll close this issue.

closed 2016-07-16 by DouglasBair

Chris--A, 2016-07-16:
You would really buy an extra bit of kit to avoid changing one line?

As I said this library will not change, you can modify the `IDLE` in this library to whatever you like, and be sure that it'll never break due to library upgrades. If an upgrade does eventuate, you aren't required to upgrade. 

But if you want the other shield, go for it.

DouglasBair, 2016-07-16:
Chris, thanks again for your support and advice.
 
Yes, going another route would avoid;
1. The current frustrations
2. problems I've found by testing the IDEs since 1.6.6.
3. Concern that if I ever (foolishly) upgrade to another level of IDE
    that I would encounter more problems.
 
----------------------------------------------------------------------------
-----
A problem in 1.6.8 is:
Note that Arduino IDE v1.6.8 fails to honor a break in a long line of code.
The sketch works fine in 1.6.6.
 
Example:
void readDS3231time(byte *second,
                    byte *minute,
                    byte *hour,
                    byte *dayOfWeek,
                    byte *dayOfMonth,
                    byte *month,
                    byte *year)
  {
  .....
  }
 
It will only pass if the line is like this;
void readDS3231time(byte *second,byte *minute,byte *hour,byte
*dayOfWeek,byte *dayOfMonth,byte *month,byte *year)
  {
  .....
  }
----------------------------------------------------------------------------
-----
A problem with 1.6.9 is:
1.6.9 fails with error that "Wire.h" isn't found.  The sketch works fine in
1.6.6.
----------------------------------------------------------------------------
-----
IDE 1.7.0:
 Seems to have fixed "not finding Wire.h" problem.
 Doesn't have the 'Open Recent' selection.
 Previously working skteches fail with error of "EEPROM class has no member
named 'length'".
 I have a logging function depending upon the feature.  I guess no one
tested for backward compatibility.
----------------------------------------------------------------------------
-----
 
I'm still testing the IDEs after 1.7.1, trying to see if I can use 1.7.10.
 
Doug
  _____  

be fixed! #4858 (#4)

<https://github.com/Chris--A/Keypad/issues/4#issuecomment-233132806> it on
GitHub, or mute
ymsL81mks5qWOoHgaJpZM4IOUJj> the thread.
<https://github.com/notifications/beacon/ARmIyJH91Evuoef6sDN3E719l7ujtj4Iks5
qWOoHgaJpZM4IOUJj.gif> 

  _____  

Version: 2016.0.7640 / Virus Database: 4627/12623 - Release Date: 07/15/16

Chris--A, 2016-07-16:
If you upgrade your IDE, this library doesn't change as it is one you have to download through the board manager. The GSM library might.

Remember the Keypad library is not maintained by Arduino.

If the errors you experienced in 1.6.6 are runtime errors, after coming from IDE 1.5.7 or below, then it is possible you have a logic error. There is a newer compiler which is less forgiving to mistakes.

DouglasBair, 2016-07-16:
Thanks for the reminder that the keypad lib isn't Arduino's, that was a
mistake on my part.
As to errors in 1.6.6, I have everything working in 1.6.6 except the lib
conflict.
It is after that (see previous list/email) that things have changed.
 
I started using Arduino in May of 2015 with 1.6.4.
 
The errors I have mentioned are compile errors, not run time.
 
Like my son says "never upgrade, it just causes problems".
 

  _____  

be fixed! #4858 (#4)

yHg1yHcks5qWQOkgaJpZM4IOUJj> the thread.
<https://github.com/notifications/beacon/ARmIyDDNwWVRwk8OOiChL2xjZ8bCy162ks5
qWQOkgaJpZM4IOUJj.gif> 

  _____  

Version: 2016.0.7640 / Virus Database: 4627/12626 - Release Date: 07/16/16