Chris--A/Keypad #13: This library was already on Github

softegg, 2017-06-28:
This library already lives on Github here:
https://github.com/Nullkraft/Keypad

Your fork is missing some of the later revisions I made for handling multiple keys down for real-time stuff by adding state information to the event handler.

DRSDavidSoft, 2017-11-01:
@softegg I think the version at @Nullkraft should be merged here, because:
1. This repo has (more) active development
2. It's linked on the official Arduino page

I'll create a pull request, if @Chris--A approves.

Chris--A, 2017-12-21:
@DRSDavidSoft, go for it. This repo is here for the community.

jamesmyatt, 2020-06-01:
This repo appears to be inactive (last commit was 5 years ago). Maybe it's better to use the Nullkraft repo now, which is ahead of this one anyway.

Nullkraft, 2020-06-02:
This existence of this repo is partly my fault. Early on, when the Arduino teams were fighting internally, they locked the playground for quite some time. I ended up supporting requests for help directly through email so it made it seem as though there was no activity. I admit that I hoped this repo could become the default because I didn't want to fight for control over the URL link used by the Arduino Library Manager.

Chris--A, please let me know if you are against updating the Arduino Library Manager to point to my Nullkraft repo.

DRSDavidSoft, 2020-06-02:
@Nullkraft Glad that the internal fight is now long over. Did you develop this library? Or rather maintain it?

Will you be willing to merge pull requests in here:
https://github.com/Chris--A/Keypad/pulls

And also some commits that @Chris--A initially did here?
https://github.com/Chris--A/Keypad/commits/master

Nullkraft, 2020-06-02:
If I understand correctly you are asking if I would be willing to accept the same pull requests at [Nullkraft/Keypad](https://github.com/Nullkraft/Keypad)? If so then yes, I never stopped supporting the Keypad library.
I was the original author of the library but I only took it as far as the state-machine control. Alexander, a real programmer, later added the keypad event routines among other things. Alexander moved on to working with Wiring which is the upstream language that Arduino IDE uses.