Chris--A/Keypad #19: Column3 recognized as Column1, too (Column-Order)

ingof, 2018-04-02:
If i use the column in ascending order the column3 will be recognized as Cloumn1, too.
In Descending Order it works fine. I used it with Arduino Uno (Actual GUI 1.8.5)

Wont work:
`byte colPins[COLS] = {2, 3, 4};`

Will work:
`byte colPins[COLS] = {4, 3, 2};`

My config for the rows:
`byte rowPins[ROWS] = {5, 6, 7, 8};`

Is this a bug?
What am i doing wrong?

tvwerkhoven, 2022-02-06:
I have the same issue with an Arduino Pro Mini 5V clone. Ordering by descending seems easy enough, thanks for solving :)

Puzzledbrain101, 2024-07-23:
// Private : Hardware scan
void Keypad::scanKeys() {
	// Re-intialize the row pins. Allows sharing these pins with other hardware.
	for (byte r=0; r<sizeKpd.rows; r++) {
		pin_mode(rowPins[r],INPUT_PULLUP);
	}

	// bitMap stores ALL the keys that are being pressed.
	for (byte c=0; c<sizeKpd.columns; c++) {
		pin_mode(columnPins[c],OUTPUT);
		pin_write(columnPins[c], LOW);	// Begin column pulse output.
		for (byte r=0; r<sizeKpd.rows; r++) {
			bitWrite(bitMap[r], c, !pin_read(rowPins[r]));  // keypress is active low so invert to high.
		}
		// Set pin to high impedance input. Effectively ends column pulse.
		pin_write(columnPins[c],HIGH);
		pin_mode(columnPins[c],INPUT);
	}
}
Hey, it seems it is because of this code by the devs that it is taking descending order. Let us say it s a 4*4 matrix so the thing is the row 0 (row 0) is connected to my pin 2, row 1 (row 2) is connected to pin 3 ... and the same goes for the column. 
MAIN ISSUE:- the thing is that you might have written it the opposite way such that
                             here 2 is row 0, 3 is row 1 and 4 is row 2. But you might have taken 4 as row 0 , 3 as row 1   and 2 as row 2 .
byte colPins[COLS] = {2, 3, 4};
that is why it is working in just one way it is not because of ascending or descending order that is not correct.