Chris--A/Keypad #21: Support to STM32F103C8T6

carloshcarvalho, 2018-06-17:
Hello, I would like to know if it is possible for this library to work with the stm32f103c8t6 cards, currently when I try to compile I get an error, and I can not change the library to work.

Voha888, 2020-03-24:
Try thiis config:

_[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
board_build.mcu = stm32f103c8t6
board_build.f_cpu = 72000000L
framework = arduino
monitor_speed = 115200_

qori-aziz, 2020-05-16:
Hello, I am currently experiencing the same issue with STM32F4. I believe it is because the different ways on handling the pins

> Try thiis config:
> 
> _[env:bluepill_f103c8] platform = ststm32 board = bluepill_f103c8 board_build.mcu = stm32f103c8t6 board_build.f_cpu = 72000000L framework = arduino monitor_speed = 115200_

What do you mean by this config? Where I can insert it?
In my implementation, the error is like this:
```
C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h:41:2: warning: #warning "Using  pinMode() INPUT_PULLUP AVR emulation" [-Wcpp] `

#warning "Using  pinMode() INPUT_PULLUP AVR emulation"

 ^

C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h: In member function 'virtual void Keypad::pin_mode(byte, byte)':

C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h:50:22: error: invalid conversion from 'byte {aka unsigned char}' to 'WiringPinMode' [-fpermissive]

  pinMode(_pin, _mode);  \

                     ^

C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h:43:30: note: in expansion of macro '_mypinMode'

#define pinMode(_pin, _mode) _mypinMode(_pin, _mode)

                             ^

C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h:80:50: note: in expansion of macro 'pinMode'

 virtual void pin_mode(byte pinNum, byte mode) { pinMode(pinNum, mode); }

                                                 ^

In file included from C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\stm32duino\hardware\STM32F4\2020.4.23\cores\maple/wirish.h:41:0,

                from C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\stm32duino\hardware\STM32F4\2020.4.23\cores\maple/WProgram.h:30,

                from C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\stm32duino\hardware\STM32F4\2020.4.23\system/libmaple/Arduino.h:31,

                from sketch\master.ino.cpp:1:

C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\stm32duino\hardware\STM32F4\2020.4.23\cores\maple/io.h:114:6: error:   initializing argument 2 of 'void pinMode(uint8, WiringPinMode)' [-fpermissive]

void pinMode(uint8 pin, WiringPinMode mode);`
```
Any help will be appreciated in tackling this error.

Voha888, 2020-05-16:
> What do you mean by this config?
Upgrade Your level to VScode+Platformio: https://platformio.org/
This config for **platformio.ini** file

Second way is: check this libraries in your **classic arduino IDE**:
[Keypad-master_2.zip](https://github.com/Chris--A/Keypad/files/4638619/Keypad-master_2.zip)
[Keyboard.zip](https://github.com/Chris--A/Keypad/files/4638623/Keyboard.zip)
[STM32-master-Keypad.zip](https://github.com/Chris--A/Keypad/files/4638617/STM32-master-Keypad.zip)