Chris--A/Keypad #25: would it be posible to change the char to int to allow more keys?

garyocampo, 2018-11-16:


brorgustav, 2019-09-11:
Did you try to redefine LIST_MAX? It controls the scanning

jtornero, 2020-05-07:
Hello @garyocampo @brorgustav @Chris--A @PaulStoffregen  

I do drive a 132 keys keyboard (but using only 121) with this library. AFAIK, I had to modify 

MAPSIZE 10 directive to MAPSIZE 12  in keypad.h to make possible to do the scanning properly. 

If I can recall, number of columns is limited to 16, so the map size is by default 16x10 keys.
But in the end it depends on your keyboard matrix. In my case, despite the theoretical size of the keys is 16x10=160 keys, my matrix is 12 columns by 11 rows, so the scanning fails. Rising the column number to 12 covers all the  keys and the keypad works properly.

There is a limit, however, in how many keys you can drive, and it is limited to Arduino digital i/o pins. I use arduino MEGA so if I call recall a maximum of a 15x15 matrix could be used. Again, it depend on how many REAL digital I/O pins you have in your design, because for instance using ethernet shield some digital pins (51?) can't be used.

And about your suggestion, it would be sure and I beg some real Arduino programmer tackle it!!!
For me, the real nice thing should be to define the matrix with strings instead of chars, so you could code your keyboard better.

 But because the library asks for CHAR, arduino accepts the ascii code for the chars, so in fact you have 255 possible keys!!! Take a look to my keyboard matrix:

```
/* THIS ARRANGEMENT MAKES POSSIBLE USING ETHERNET SHIELD
//BECAUSE PINS 50-53 ARE RESERVED IN ARDUINO MEGA FOR THAT SHIELD*/
char hexaKeys[ROWS][COLS] = {
///*COL ------->  1   2   3   4   5   6   7   8   9  10  11  12*/
///*ROW |  PIN    25  27  29  31  33  35  37  39  41  43  45  47*/
/* 1  |  24*/ {  1,  2,  3,  4,  5,  0,  0,  0,  0,  0,  0,  0},
/* 2  |  26*/ {  6,  7,  8,  9, 10, 11, 12, 13, 14, 15, 16, 17}, /*OK
/* 3  |  28*/ { 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29}, /*OK
/* 4  |  30*/ { 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41}, /*OK
/* 5  |  32*/ { 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53}, /*OK
/* 6  |  34*/ { 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64,  0}, /*OK
/* 7  |  36*/ {  0, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74,  0}, /*OK
/* 8  |  38*/ { 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86}, /*OK
/* 9  |  40*/ { 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98}, /*OK
/* 10 |  42*/ { 99,100,101,102,103,104,105,106,107,108,109,110}, /*OK
/* 11 |  44*/ {111,112,113,114,115,116,117,118,119,120,121,122}}; /*OK*/

byte rowPins[ROWS] = {22,24,26,28,30,32,34,36,38,40,42}; // Row pins connection
byte colPins[COLS] = {23,25,27,29,31,33,35,37,39,41,43,45}; //Column pins connection

String measures[]={
"NIMP","UND","FEM","DEL","FUN","MALE","600","590","580","570","560","550",
"540","530","520","510","500","490","480","470","460","450","440",
"430","420","410","400","390","380","370","360","350","340","330",
"320","310","300","290","280","270","260","250","240","230","220",
"210","200","190","180","170","160","150","140","130","120","110",
"100","90","80","70","60","50","40","30","20","115","105","95",
"85","75","65","55","45","35","25","125","235","225","215","205",
"195","185","175","165","155","145","135","245","355","345","335",
"325","315","305","295","285","275","265","255","365","475","465",
"455","445","435","425","415","405","395","385","375","485","595",
"585","575","565","555","545","535","525","515","505","495"};
```

What I do next is "recover" the measure string with the index of the pressed key.

Letting the keyboard to return the string directly instead of a char or the integer would be absolutely cool. But that's another story!!!

By the way, what's your keyboard for? I'm so curious!!

All the best

Jorge