Chris--A/Keypad #41: RaspberryPi Pico - adding library kills I/O pullups

david284, 2022-03-14:
I have a raspberryPi Pico (based on RP2040), and when I just include the library (i.e. don't reference any functions or define any I/O's), I/O that have been defined as pull-ups no longer have a pull-up
for example
in setup(), I have
`pinMode(14, INPUT_PULLUP);`

So,  before adding `#include <keypad.h>`
GP14 reads  a steady 3.2v with a multimeter
after adding just `#include <keypad.h>`
GP14 now reads a varying value around 140mV

It's affecting the actual pins for the key matrix, as I first noticed that when I had the full code for one of the examples, just touching the pins caused key presses to be registered - a sure sign of lack of pull-ups

I don't understand why it should be affecting other IO pins not used by the keypad library, but it seems to affect them all
I realise I could add pullups to the actual kaypad matrix, but more worried that it affects other pins that I intend to use for other features that rely on internal pullups working

My apologies if I've missed something obvious

I'm using PlatformIO inside VSCode if that makes any difference
and have
`lib_deps = chris--a/Keypad@^3.1.1` in platformio.ini

My entire program is below, with the majority commented out

```
#include <Arduino.h>
#include <Keypad.h>

/*
const byte ROWS = 4; //four rows
const byte COLS = 3; //three columns
char keys[ROWS][COLS] = {
  {'1','2','3'},
  {'4','5','6'},
  {'7','8','9'},
  {'#','0','*'}
};
byte rowPins[ROWS] = {6, 7, 8, 9}; //connect to the row pinouts of the keypad
byte colPins[COLS] = {10, 11, 12}; //connect to the column pinouts of the keypad

Keypad keypad = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );
*/

void setup(){
  Serial.begin(115200);
    pinMode(14, INPUT_PULLUP);
    pinMode(15, INPUT_PULLUP);
}

void loop(){
/*
  char key = keypad.getKey();

  if (key != NO_KEY){
    Serial.println(key);
  }

  */
}

```

Koepel, 2023-04-14:
I can confirm that the Keypad library does not work with a Raspberry Pi Pico, and also kills I/O pullups.

The Raspberry Pi Pico in normal Arduino mode is on top of Mbed.
The definition of INPUT_PULLUP is removed, and the INPUT_PULLUP is placed in a enum.
See: https://github.com/arduino/ArduinoCore-mbed/blob/main/variants/RASPBERRY_PI_PICO/pinmode_arduino.h

In case the INPUT_PULLUP was missing, a fix was added in Keypad.h: https://github.com/Chris--A/Keypad/blob/master/src/Keypad.h
The fix is from 2013. It brings back the definition for INPUT_PULLUP which is wrong for a Raspberry Pi Pico, then it does not use a `pinMode(_pin,INPUT_PULLUP);` but a `pinMode(_pin, INPUT); digitalWrite(_pin, 1);` which does not result into a pullup for the pin.

hacksterous, 2023-08-22:
After locally changing that line to `pinMode(_pin,INPUT_PULLUP);` instead of `pinMode(_pin, INPUT); digitalWrite(_pin, 1);` the keypad works fine on the Pico.