Chris--A/Keypad #46: The readings are unstable on the Uno R4 board.

eMUQI, 2023-07-11:
Has anyone tried using this library on the Uno R4 board? I attempted to use the code below, but the data obtained was very unstable and the serial monitor kept printing 1. Although there was a response when pressing the button, it cannot be used for practical applications on the R4 board.

```
// Include the Keypad library
#include <Keypad.h>

// Define the number of rows and columns in the keypad
const byte ROWS = 4;
const byte COLS = 4;

// Define the keys on the keypad
char keys[ROWS][COLS] = {
  { '1', '2', '3', 'A' },
  { '4', '5', '6', 'B' },
  { '7', '8', '9', 'C' },
  { '*', '0', '#', 'D' }
};

// Define the pins connected to the rows and columns of the keypad

byte rowPins[ROWS] = { 9, 8, 7, 6 };
byte colPins[COLS] = { 5, 4, 3, 2 };

//Create an object of keypad
Keypad keypad = Keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS);

void setup() {
  Serial.begin(9600);
}

void loop() {
  char key = keypad.getKey();  // Read the key

  // If a key is pressed, print the key on the serial monitor
  if (key) {
    Serial.print("Key Pressed : ");
    Serial.println(key);
  }
}
```

[image]

nyk-yeon, 2023-07-28:
I also had this problem, and when I implemented and tested the scanKeys() function of the library directly in main.c, the problem disappeared.

However, when we added #include <Keypad.h>, the same problem occurred, and it is confirmed that the problem occurred in the macro part of the header below.
`
Keypad.h

// bperrybap - Thanks for a well reasoned argument and the following macro(s).
// See http://arduino.cc/forum/index.php/topic,142041.msg1069480.html#msg1069480
#ifndef INPUT_PULLUP
#warning "Using  pinMode() INPUT_PULLUP AVR emulation"
#define INPUT_PULLUP 0x2
**// #define pinMode(_pin, _mode) _mypinMode(_pin, _mode)    <<--- UNO R4 error**
#define _mypinMode(_pin, _mode)  \
do {							 \
	if(_mode == INPUT_PULLUP)	 \
		pinMode(_pin, INPUT);	 \
		digitalWrite(_pin, 1);	 \
	if(_mode != INPUT_PULLUP)	 \
		pinMode(_pin, _mode);	 \
}while(0)
#endif

`
When I modified it as above, it worked normally.

The above macro is not working properly on UNO R4, and it seems to interfere with pull-up functionality.

M4c3j, 2023-09-21:
Hello,

I have the same issue with Giga but there was some helpfull info:
Change Keypad.h:
#ifndef INPUT_PULLUP
to:
#if ! (defined(INPUT_PULLUP) || (ARDUINO_API_VERSION >= 10000))
It works for me.

paulwitschey, 2024-05-05:
> I also had this problem, and when I implemented and tested the scanKeys() function of the library directly in main.c, the problem disappeared.
> 
> However, when we added #include <Keypad.h>, the same problem occurred, and it is confirmed that the problem occurred in the macro part of the header below. ` Keypad.h
> 
> // bperrybap - Thanks for a well reasoned argument and the following macro(s). // See http://arduino.cc/forum/index.php/topic,142041.msg1069480.html#msg1069480 #ifndef INPUT_PULLUP #warning "Using pinMode() INPUT_PULLUP AVR emulation" #define INPUT_PULLUP 0x2 **// #define pinMode(_pin, _mode) _mypinMode(_pin, _mode) <<--- UNO R4 error** #define _mypinMode(_pin, _mode) do { if(_mode == INPUT_PULLUP) pinMode(_pin, INPUT); digitalWrite(_pin, 1); if(_mode != INPUT_PULLUP) pinMode(_pin, _mode); }while(0) #endif
> 
> ` When I modified it as above, it worked normally. The above macro is not working properly on UNO R4, and it seems to interfere with pull-up functionality.
> 
> +If the error still occurs, you need to comment out the entire macro.

Thanks @nyk-yeon, this solved the problem I had with the R4 Wifi